ros/ros_tutorials #20: I suggest you to change the node names.

robotpilot, 2015-02-03:
$ rosrun turtlesim turtlesim_node 
$ rosrun turtlesim turtle_teleop_key 

$ rosnode list
/rosout
/teleop_turtle
/turtlesim 

We usually use the commands above. I think that a beginner confuse the node names, because executable node name and real name are different. The turtlesim package is one of the tutorials for beginner. So,  I suggest you to change the node names. How about that?

1. turtlesim_node VS turtlesim
2. turtle_teleop_key VS teleop_turtle

ros_tutorials/turtlesim/tutorials/teleop_turtle_key.cpp 
......
ros::init(argc, argv, "teleop_turtle");
.....

dirk-thomas, 2015-02-03:
Currently the tutorial works with any ROS distro. Changing the naming would require the tutorial to be distro specific.

I am not sure if the small renaming justifies the additional user confusion of having separate tutorials. May be a note in the tutorial mentioning the mapping would be less disruptive.

robotpilot, 2015-02-04:
I heard above question from student in a brief question-and-answer period, courses.
but, your comment is reasonable opinion. I agree your comment.
Thanks dirk-thomas.

closed 2015-02-04 by tfoote