ros/ros_tutorials #24: Turtlesim /spawn service uses different coordinate system

jarvisschultz, 2015-06-12:
The numbers used when spawning a turtle via the `/spawn` service are sent directly to the `turtlesim::Turtle` constructor in Qt coordinates. All other numbers used in the various topics and services are in a different coordinate system; e.g., `/turtleX/pose` topics and `/turtleX/teleport_absolute` services. What this means is that teleporting a turtle to a given (x,y) and spawning a turtle at the same (x,y) will result in turtles at two different locations.

See [this answers.ros.org question](http://answers.ros.org/question/211156/question-on-frame-transform/?answer=211208).

dirk-thomas, 2015-08-20:
Thank you for paying attention to details and for the pull request.

closed 2015-08-20 by dirk-thomas

addy1997, 2020-12-01:
how to spawn the turtle at random locations?

jarvisschultz, 2020-12-01:
@addy1997 this question would be better asked on [answers.ros.org](https://answers.ros.org/) There are many more people reading those questions and trying to help people than in bug reports closed 5 years ago. You're more likely to get the help that you need, and any help you do receive is more likely to be helpful to others. Best of luck!