ros/ros_tutorials #30: rosrun turtlesim turtlesim_node segmentation fault

Ghinda94, 2016-03-29:
I am new to ROS. I am working through the beginner's tutorials. Version: ROS Indigo, OS Kubuntu 14.04. I was trying rosrun turtlesim turtlesim_node and the result was: Segmentation fault ( core dumped). What can I do? Thanks in advance.

dirk-thomas, 2016-03-29:
Please ask the question on answers.ros.org instead. Also provide more information on the exact steps you did, share your custom sources as well as try running it in gdb to get more information about the segfault.

dirk-thomas, 2016-03-29:
You can post the link to answers here for the record and close the ticket when it's answered.

dirk-thomas, 2016-03-29:
Since the question was already asked before filling this ticket (http://answers.ros.org/question/230401/rosrun-turtlesim-turtlesim_node-segmentation-fault/) I will close this for now. If it turns out to be a problem in `ros_tutorials` it can be reopened.

closed 2016-03-29 by dirk-thomas