ros/ros_tutorials #32: Why "subs_.push_back" in listener_with_userdata.cpp

weiweikong, 2016-05-17:
1. What is the meaing for `push_back()` here and why we use vector to manage the construction of the three different `chatterCallback`?

2. Why can not use the bellowing codes
```
void init()
  {
    node_handle_.subscribe<std_msgs::String>("chatter", 1000, boost::bind(&Listener::chatterCallback, this, _1, "User 1"));
    node_handle_.subscribe<std_msgs::String>("chatter", 1000, boost::bind(&Listener::chatterCallback, this, _1, "User 2"));
    node_handle_.subscribe<std_msgs::String>("chatter", 1000, boost::bind(&Listener::chatterCallback, this, _1, "User 3"));
  }
```
to initialse the different callback function?

dirk-thomas, 2016-05-17:
The `subscribe` function returns a reference counted publisher. When all copies of it go out of scope it will unsubscribe from this topic. Therefore the subscribers are stored in a vector to prevent them from going out of scope.

Please read the documentation of the function (`NodeHandle::subscribe`: http://docs.ros.org/indigo/api/roscpp/html/classros_1_1NodeHandle.html#a317fe4c05919e0bf3fb5162ccb2f7c28) and ask future questions on answers.ros.org.

closed 2016-05-17 by dirk-thomas

weiweikong, 2016-05-18:
@dirk-thomas thanks for your kind reply. There are still some confusions. More details please refer to http://answers.ros.org/question/234538/why-subs_push_back-in-listener_with_userdatacpp .