ros/ros_tutorials #33: Command vel genrated  using teleop_twist_keyboard stops turtle but not robot

atuldiv, 2016-07-22:
Hello,

I have one issue in ROS
Using teleop_twist_keyboard we are generating cmd_vel messages(geometry_msgs/Twist) and this message are send to turtlesim_node where the turtle moves successfully.But if I suddenly stops sending command then the turtle stops moving .

Now similarly If I use teleop_twist_keyboard for my robot , I am sending cmd_vel to my robot then it moves successfully But If I suddenly stops sending command still the robots receive message and it is moving .
Please can anyone suggest what is difference and issue in this.

dirk-thomas, 2016-07-22:
Please ask questions on http://answers.ros.org

closed 2016-07-22 by dirk-thomas

atuldiv, 2016-07-25:
Hello,

Yes I will definitely ask questions on given link.

Thanks and regards
 - Atul Divekar
   09970588826