ros/ros_tutorials #38: does node mimic really publish the topic /output subscribe the topic /input?

ly11tea, 2017-02-09:
In the document [here](http://wiki.ros.org/turtlesim),node mimic's Subscribed Topic is /input,Published topic is /output ,
but when i read the source code of mimic ,i found that node mimic subscribes the topic  input/pose, Publishes the topic output/cmd_vel.     where:
```c++
  ros::NodeHandle input_nh("input");
  ros::NodeHandle output_nh("output");
  twist_pub_ = output_nh.advertise<geometry_msgs::Twist>("cmd_vel", 1);
  pose_sub_ = input_nh.subscribe<turtlesim::Pose>("pose", 1, &Mimic::poseCallback, this);
```
it could be confusing ,because if i remap the topic in the launch file i will Write:
```xml
    <remap from="input" to="turtlesim1/turtle1/pose"/>
    <remap from="output" to="turtlesim2/turtle1/cmd_vel"/>
```
but in the end,mimic code subscribes the topic  **turtlesim1/turtle1/pose/pose**,Publishes the topic **turtlesim2/turtle1/cmd_vel/cmd_vel**
obviously ,it doesn't  work
i found one tutorial [here](http://wiki.ros.org/ROS/Tutorials/UsingRqtconsoleRoslaunch) ,the corresponding part of launch file is:
```xml
    <remap from="input" to="turtlesim1/turtle1"/>
    <remap from="output" to="turtlesim2/turtle1"/>
```
of course it works,but it seems to be contradict to the document in the beginning of this issue ,where:
Subscribed Topic is /input(**message type is turtlesim/Pose**),Published topic  is /output(**message type is geometry_msgs/Twist**) .because turtlesim2/turtle1 and turtlesim1/turtle1 is node here.
do i misunderstand something ?thanks a lot!

dirk-thomas, 2017-02-14:
I am not sure I understand your point. For each turtle there are topics `pose` and `cmd_vel` which are in the namespace of the turtle (e.g. `turtleX`). The same for the `mimic` node - it has two topics within the node names `input` and `output`.

Therefore you can either:
* remap the node namespace: `input` -> `turtlesim1/turtleN` and `output` -> `turtlesim1/turtleN`
* or remap the specific topics: `input/pose` -> `turtlesim1/turtleN/pose` and `output/cmd_vel` -> `turtlesim1/turtleN/cmd_vel`

If you think the documentation can be improved please proposed a concrete diff to the existing wiki page. I can't propose any change since I don't understand your confusion I guess.

dirk-thomas, 2017-03-06:
I will close this for now due to no response. Please feel free to comment here with additional information and the ticket can be reopened.

closed 2017-03-06 by dirk-thomas

weidongguo, 2017-11-10:
I had the same confusion when I read the tutorial <http://wiki.ros.org/ROS/Tutorials/UsingRqtconsoleRoslaunch>

I think of `input `and `output` as topics in the beginning and that confuses me.
Now I think of them as namespaces, then it makes more sense.

remap will map namspace `input `to namespace `turtlesim1/turtle1`. Topic names within `input` remain the same
for example,
`pose` happens to be a topic within namespace `input`, after the mapping, it will have a namespace `turtlesim1/turtle1`, and the _full_ topic name now is `turtlesim1/turtle1/pose`

The **benefit** of this is that you don't have to explicitly state all the topic names if the only change is just the namespace but not the local topic names.

On the other hand, the actual topics are `input/pose`, and `output/cmd_vel`,
so it also make sense to remap `input/pose` -> `tutorialsim1/turtle1/pose`
and `output/cmd_vel` -> `tutorialsim2/turtle1/cmd_vel`
because we want mimic to listen to topic  `tutorialsim1/turtle1/pose` and publish whatever it's received to `tutorialsim2/turtle1/cmd_vel`.

However, we have to explicitly write out the topic names that are the same. Imagine if you have 100 topics. you don't wanna type all that out. we are coder. we want to do things in an automatic fashion.

I hope this helps.

dirk-thomas, 2017-12-13:
@weidongguo Please consider to contribute to the wiki and improve the content you find confusing.

RicardoDanielCPinto, 2019-07-17:
Hey,
I found this issue because i got the same confusion and i didn't got the answer here but i did found it by my self.
What he's trying to say is that the input topic which is turtlesim1/turtle1/pose seems to be a position... "Pose"... But its not if we check the type, it's "turtlesim/Pose" and if we check the data structure of this type we get:
float32 x
float32 y
float32 theta
float32 linear_velocity    <------------- 
float32 angular_velocity <-------------

So its confusing trying to mimic a velocity with a position, but what happens is that mimic is getting the velocity information from /Pose topic to publish on turtlesim2/turtle1/cmd_vel

Check the source of mimic here: http://docs.ros.org/diamondback/api/turtlesim/html/mimic_8cpp_source.html

If you think its worth a change on wiki, please do it for me.
Sorry my english