ros/ros_tutorials #40: keyboard cannot work when running "rosrun turtlesim turtle_teleop_key"

guchaojie, 2017-06-09:
My enviroment is 
NAME="Ubuntu"
VERSION="16.04.2 LTS (Xenial Xerus)"

kernel is  4.4.0-75-generic 
 running step is:
1. roscore
2. rosrun  rosrun turtlesim turtlesim_node
3. rosrun turtlesim turtle_teleop_key
The result is :  keyboard cannot work to drive turtlebot move in simulator
 the root cause is 
88     if(read(kfd, &c, 1) < 0)
 89     {
 91       perror("read():");
 92       exit(-1);
 93     }
   line 88:  read function is being hung there, cannot running down to publish turtle1/cmd_vel topic.

dirk-thomas, 2017-06-09:
I am not sure I understand your problem. The line containing the `read` is waiting for keyboard events. It is supposed to block until you press any key, then the loop processes the event and waits for the next input. Please describe your problem with more details: what are the exact steps you do, what is the exact output, what are expecting, etc?

guchaojie, 2017-06-12:
I found the problem reason is that I put the turtlesim simulator in front of screen to enter keyboard,  the right way is to put terminal in front of screen as active to wait for keyboard event

closed 2017-06-12 by guchaojie