ros/ros_tutorials #47: Unexpected move of the simulated turtle for large linear velocity

zhoulaifu, 2018-11-26:
The simulated turtle moves in a wrong direction if the linear velocity  is large. To reproduce this issue, try:

`rostopic pub /turtle1/cmd_vel geometry_msgs/Twist '[10000000000000000,0,0]' '[0,0,0]'`

The simulated turtle should move horizontally in the case above, which is not the case from the current version:

[image]

There is an easy fix to this problem if this issue is to be addressed.

betab0t, 2018-12-25:
seems like this behavior occurs also with smaller linear velocity values, the y-axis change is just too small to notice. looks like thats the root cause of this issue: https://github.com/ros/ros_tutorials/blob/2b1de43ba413bf59bbb7eb0578bf79e1721049de/turtlesim/src/turtle.cpp#L158
so as you can see even if orient_ is zero, still there will be movement on the y-axis. ill try to provide a PR soon...

dirk-thomas, 2019-02-04:
While the drawn line isn't very intuitive I don't think the result is wrong. Whenever the turtle hits the limit of the screen the position is clamped back into the visible range. In this case the distance is extremely large in x direction and due to floating point inaccuracy significant enough in y direction to travel to the border of the screen.

Since `turtlesim` is targeted to be a simple tutorial of ROS functionality I don't think it needs to handle this extreme input data anyhow different.

Therefore I will go ahead and close this ticket for now. Please feel free to continue commenting on it.

closed 2019-02-04 by dirk-thomas

zhoulaifu, 2019-02-05:
I believe that the FP inaccuracy can be largely reduced if, in ros_tutorials/turtlesim/src/turtle.cpp, we change

````
  pos_.rx() += std::sin(orient_ + PI/2.0) * lin_vel_ * dt;
  pos_.ry() += std::cos(orient_ + PI/2.0) * lin_vel_ * dt;
````
to 
````
  pos_.rx() += std::cos(orient_) * lin_vel_ * dt;
  pos_.ry() += - std::sin(orient_) * lin_vel_ * dt;
````
This is based on the mathematical facts that sin(x+pi/2)=cos(x) and cos(x+pi/2)=-sin(x).    See https://en.wikipedia.org/wiki/List_of_trigonometric_identities for example.

dirk-thomas, 2019-02-05:
The proposed patch will indeed make the starting configuration better. But for any move with extreme speeds after a rotation the problem stays the same, see e.g. the result of the following sequence:

* `turtle_teleop_key`: once forward
* `turtle_teleop_key`: once left turn
* `turtle_teleop_key`: once forward
* `rostopic pub /turtle1/cmd_vel geometry_msgs/Twist '[10000000000000000,0,0]' '[0,0,0]'`

[image]

You would expect the last segment of the drawn line to be in the same direction as the second last segement.

If you would like to create a PR with your proposed change I am happy to merge it. I just don't see the point if it only addresses one special case.

zhoulaifu, 2019-02-10:
The screenshot you showed  above (the one with `turtle_teleop_key`) should be as expected (and therefore, should not invalidate the proposed fix IMHO). If you review line160-168 of ros_tutorials/turtlesim/src/turtle.cpp:

```
  // Clamp to screen size
  if (pos_.x() < 0 || pos_.x() > canvas_width ||
      pos_.y() < 0 || pos_.y() > canvas_height)
  {
    ROS_WARN("Oh no! I hit the wall! (Clamping from [x=%f, y=%f])", pos_.x(), pos_.y());
  }

  pos_.setX(std::min(std::max(static_cast<double>(pos_.x()), 0.0), static_cast<double>(canvas_width)));
  pos_.setY(std::min(std::max(static_cast<double>(pos_.y()), 0.0), static_cast<double>(canvas_height)));
```

To be sure, you can try commenting out the two lines with pos_.setX and pos_setY. I got this screenshot with your sequences of commands above:
[image]