ros/ros_tutorials #48: Data races in the turtlesim node

zhoulaifu, 2018-12-14:
I am testing the package with ThreadSanitizer, aka TSan, a GCC/Clang feature for detecting data races. Below is one (among many) data race detected by TSan. The location of the error is shown (using -g option). 

To reproduce the issue, compile the package with "-fsanitize=thread -g", launch turtlesim_node, and then you will see the data race warning. 

```
WARNING: ThreadSanitizer: data race (pid=20894)
  Write of size 8 at 0x7d1800017fa0 by main thread:
    #0 free <null> (libtsan.so.0+0x000000025819)
    #1 <null> <null> (libglib-2.0.so.0+0x000000047289)
    #2 main /home/zhfu/catkin_ws/src/turtlesim_official/src/turtlesim.cpp:60 (turtlesim_official_turtlesim_node+0x00000041d178)

  Previous write of size 8 at 0x7d1800017fa0 by thread T5:
    #0 calloc <null> (libtsan.so.0+0x00000002565d)
    #1 g_malloc0 <null> (libglib-2.0.so.0+0x00000004f810)

  Thread T5 'pool' (tid=20908, running) created by main thread at:
    #0 pthread_create <null> (libtsan.so.0+0x000000027577)
    #1 <null> <null> (libglib-2.0.so.0+0x00000008e8ff)
    #2 main /home/zhfu/catkin_ws/src/turtlesim_official/src/turtlesim.cpp:59 (turtlesim_official_turtlesim_node+0x00000041d173)

SUMMARY: ThreadSanitizer: data race ??:0 __interceptor_free

```

dirk-thomas, 2018-12-26:
The two lines seem to point to the `QApplication` constructor and its `exec` function. I don't see what this package should do differently. Please consider to provide a pull request.

dirk-thomas, 2019-01-28:
I will go ahead and close the ticket due to no response. Please feel free to comment with the requested information and the ticket can be reopened.

closed 2019-01-28 by dirk-thomas