ros/ros_tutorials #61: Add actions to turtlesim

maryaB-osr, 2019-09-26:
Turtlesim doesn't have any actions, but I think it's important for the sake of the tutorials that it include at least one simple action. Actions are one of the core concepts that need to be introduced, and it would damage the tutorial set if we had to switch to another example just for the sake of explaining actions. 

One simple action with some kind of feedback that users can see is all that's needed.

jacobperron, 2019-09-27:
How about an action where the goal is to rotate on the spot to face a particular direction?
The interface could be defined as follows:

```
# RotateToAngle.action

# The desired angle in degrees
float64 angle
---
# The angular displacement in degrees to the starting position
float64 angle_displacement
---
# The remaining number of degrees to rotate
float64 angle_remaining
```

dirk-thomas, 2019-09-27:
I like the rotation idea. In general since any action is running for a bit what happens with other control input during that time? Canceling the running action on control input would probably be best to also show that part of actions.

Following [REP 103](https://www.ros.org/reps/rep-0103.html) the angle should be in radians instead of degrees.

Other messages in this package use `float32` and the orientation of the turtle is described as `theta`. I would also suggest to keep the field names short:

```
# The desired heading in radian
float32 theta
---
# The angular displacement in radian to the starting position
float32 delta
---
# The remaining rotation in radian
float32 remaining
```

dirk-thomas, 2019-09-27:
Maybe name the action `RotateAbsolute.action` to match the teleportation services?

jacobperron, 2019-09-27:
@dirk-thomas All good suggestions. I'll move forward with implementation.

When another control input is received during the action, it might make more sense for the action to *abort* instead, as opposed to an explicit *cancel* input. Then the user is exposed to all possible terminal states.

jacobperron, 2019-09-27:
Action added in #62

PR #62 (2019-10-01): Add RotateAbsolute action to turtlesim
Resolves #61
closed 2019-10-02 by dirk-thomas