ros/ros_tutorials #72: ros2 param set /turtlesim background_* needs to be "triggered"

maryaB-osr, 2019-10-16:
See the [Understanding ROS 2 parameters](https://github.com/ros2/ros2_documentation/pull/347#discussion_r331814709) tutorial discussion.

Setting the background color parameter with
```
ros2 param set /turtlesim background_r 150
```
should update the turtlesim window immediately, but it needs to be triggered by calling
```
ros2 service call /clear std_srvs/srv/Empty
```
because of some leftover functionality from ROS 1.

dirk-thomas, 2019-10-17:
There is no way to change the background color without wiping all previous drawn lines. I am not sure if that is desired - to implicitly clear when changing the parameter?

maryaB-osr, 2019-10-17:
I think that's fine because now they'll have to explicitly clear anyway. I'm more concerned with the extra instructions, not what's actually happening. So if it can all be done at once (the clear and the color change) that's better.

dirk-thomas, 2019-10-17:
So just to be clear: every time a background color parameter is set the whole scene should be cleared implicitly / automatically?

maryaB-osr, 2019-10-17:
I don't think it _should_ be. If there's a way to make it happen without clearing the lines that would be best. 

But, since you said there is no way to change the color without clearing the drawn lines, I'm fine with the whole screen being cleared as long as the background changes automatically. 

I will just add a note to the tutorial that the lines will clear - that's much better than the 4 or 5 extra sentences I have now explaining how you have to call `/clear` to get the parameter setting to change.

dirk-thomas, 2019-10-18:
Early Christmas time :gift: #75 separates the background color from the painted turtle paths which makes it possible to update the former without resetting the later. Also the background is immediately redrawn when a parameter event occurs.

maryaB-osr, 2019-10-18:
Awesome! Thanks @dirk-thomas. I'll update the parameters tutorial once #75 is merged in

closed 2020-02-03 by dirk-thomas

Sridhar701Pitt, 2020-06-15:
I'm working through ros2 tutorials at [Understanding ROS 2 Parameters](https://index.ros.org/doc/ros2/Tutorials/Parameters/Understanding-ROS2-Parameters/#ros2params)
I still need to execute the `/clear` service to show the changes and it wasn't mentioned anywhere in the tutorial.

dirk-thomas, 2020-06-15:
While a new version was tagged in this repo for Dashing (https://github.com/ros/ros_tutorials/commit/de7271c6ce920f989239f57dfdb380d01ce867d3) it doesn't seem it was ever released with `bloom`: https://github.com/ros/rosdistro/blob/05e1fa9b6510c9a71a65e8d786b37934e38539b9/dashing/distribution.yaml#L3493

@claireyywang Can you clarify since you did the upstream release?

claireyywang, 2020-06-16:
@dirk-thomas  honestly I don't remember if I did since it's been a while, but I can do a release now if that's the right move.

claireyywang, 2020-06-16:
released at https://github.com/ros/rosdistro/pull/25490

PR #75 (2019-10-18): separate background color from drawn paths, trigger repaint on parameter changes
Fixes #72.