ros/ros_tutorials #86: [ROS 2] Invalid context error when stopping turtle_teleop_key

jacobperron, 2020-05-21:
Sometimes during the shutdown of turtle_teleop_key (i.e. Ctrl-C), we see the following error:

```
[ERROR] [1590100346.240527469] [rclcpp]: failed to create wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-foxy-rcl-1.0.0/src/rcl/wait.c:128
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create wait set in Executor constructor
```

Looks like an issue with the signal handling logic.

I've seen the same behavior with Fast-RTPS and CycloneDDS.

ivanpauno, 2020-05-28:
Looks like duplicate of https://github.com/ros2/rclcpp/issues/1139.

ivanpauno, 2020-06-25:
Closing in favor of https://github.com/ros2/rclcpp/issues/1139

closed 2020-06-25 by ivanpauno

jacobperron, 2020-07-10:
Note, this was a bug that came up during Foxy testing. It would be nice if a solution to ros2/rclcpp#1139 could be backported to Foxy.