ros/ros_tutorials #93: turtlesim: Support commanding the turtles with holonomic motion

Tiryoh, 2020-07-22:
I made a change to turtlesim and support commanding the turtles with holonomic motion, like an omni-directional robot.

https://github.com/Tiryoh/omni_turtlesim_ros2

[image]

At first I was going to release it as a separate package, but I was advised to integrate it into this package (https://github.com/ros/rosdistro/pull/25854).

Is it possible to integrate this feature into turtlesim? I'd like to open the PR to the following branches if this feature would be welcome.

* melodic-devel
* noetic-devel
* dashing-devel
* eloquent-devel
* foxy-devel

dirk-thomas, 2020-07-22:
`turtlesim` is meant as an example demonstrating the concepts of ROS. It tries to be simple and easy to understand. 

I don't see how strafing and changing the speed adds any value for beginners of ROS - on the opposite I think these additions would make the code more complex and difficult to understand.

What benefit would this feature bring to users when learning ROS?

@jacobperron FYI

jacobperron, 2020-07-22:
IIUC, allowing for holonomic motion in the simulator doesn't change the users experience going through the tutorials (assuming the turtlesim teleop node doesn't change; I don't think it should), but if someone wanted to command it in this way with another node they could.

I think there's value in allowing someone to experiment with their own controllers in a less constrained way. But it would be good to hear if there's a more specific use-case.

I don't think the change adds much complexity; I think this is the diff: https://github.com/Tiryoh/omni_turtlesim_ros2/commit/989cf0f0471cae7bda1128c55695540543144149

Tiryoh, 2020-07-23:
@dirk-thomas, thank you for your responding.
@jacobperron, thank you for your comment.

> IIUC, allowing for holonomic motion in the simulator doesn't change the users experience going through the tutorials (assuming the turtlesim teleop node doesn't change; I don't think it should), but if someone wanted to command it in this way with another node they could.

> I don't think the change adds much complexity; I think this is the diff: Tiryoh/omni_turtlesim_ros2@989cf0f

Yes, I don't plan to change the users experience who are learning ROS through the tutorials through this update, and I have no plan to update the turtlesim teleop node.

> `turtlesim` is meant as an example demonstrating the concepts of ROS.

I think I understand what this means.  I understood that you are arguing that the ROS tutorials with turtlesim should be easy to learn for beginners, is my understanding consistent with your opinion?

> I don't see how strafing and changing the speed adds any value for beginners of ROS - on the opposite I think these additions would make the code more complex and difficult to understand.
>
> What benefit would this feature bring to users when learning ROS?

There is a demand to use ROS with omni-directional robots.
If turtlesim can work with linear.y twist messages in addition to angular.z and linear.x, I think that satisfies the demand mentioned above.

dirk-thomas, 2020-07-23:
> I don't think the change adds much complexity; I think this is the diff: Tiryoh/omni_turtlesim_ros2@989cf0f

The screenshot indicated additional changes to the `teleop` node so I assumed those would also be part of the patch.

> I don't plan to merge this controller node to the turtlesim package.

:+1:

@Tiryoh Please go ahead and create a pull request targeting the default branch and we can iterate on the patch if necessary.

Tiryoh, 2020-07-29:
@dirk-thomas I made a Pull Request #94.  Could you check it out?

dirk-thomas, 2020-07-31:
Addressed by #94.

closed 2020-07-31 by dirk-thomas

tianb03, 2021-06-28:
I just find out that the turtle is holonomic now.  Because I am teaching ROS these years so want to share some thoughts here though this issue is closed.

Introducing the holonomic movement into Turtlesim is great, it does not change the learning experience. And I also want this feature to test some easy multi-robot formation or cooperation stuff. **However, it brings in some confusion during learning TF using turtle_tf.** My students now solve the problem mainly based on the global info, the relation between two turtles can be ignored and you can command the follower based on v.x and v.y with a zero-valued a.z. A differential model may indicate that the follower must perceive the target turtle.

Moreover, the kinematic model of the turtle is no longer the same as a Turtlebot. 

It won't be a problem during our class. I don't know how this will affect the self-learner. I think your intuition is considerable @dirk-thomas .  I prefer a default differential turtle and can be switched to omni-directional. 

best regards,

jacobperron, 2021-06-28:
@tianb03 Please avoid commenting on old and closed tickets since the comments will not be that visible. Prefer to open a new ticket (if there is not already one open). 

Having an option to disable holonomic motion sounds okay to me. I would lean towards leaving the holonomic feature on by default though, so as to not break existing "out-of-box" behavior. Please consider proposing the change as a new ticket (preferably in the form as a pull request)!