ros/ros_tutorials #99: [Enhancement] - Add a "Range color sensor"

pxalcantara, 2020-09-09:
Hi, 

I was thinking about adding a sensor to get the color ahead of the turtle as a "range sensor" but, instead of getting the distance, once the turtlesim is just a 2D simulator, get the color in a defined distance. The image below shows the idea of the behavior.  

[image]

In the image is possible to see the turtle facing different color and publishing the RGB values at the topic `turtle1/range_color_sensor` that is also, showed in the image.

The idea of this new feature is making possible implement some behaviors like "obstacle detecting" and, once the turtle is basically a differential robot, it'll be possible to use the turtlesim as a platform to demonstrate different concepts and aspects of the mobile robots.

Do you think that is an interesting feature? If so I'll proceed with the first version to the PR.

dirk-thomas, 2020-09-10:
`turtlesim` is meant to be a simple interactive tutorial to teach the ROS basics. While there are many possible features which could be added I think they would also inevitably make it more complex and less easy to understand by new ROS users. Therefore I am leaning against adding this feature idea. Instead I would suggest to create your own package if you think this would be useful to a broad audience.

pxalcantara, 2020-09-11:
Ok @dirk-thomas Thank you. I'll think in a way of, maybe, inherit from the turtlesim libraries or just use them and implement these new functionalities in another package.

closed 2020-09-11 by pxalcantara