ros/ros_tutorials #101: catkin_make error

zhoulaifu, 2020-10-09:
I cloned this repository to a local directory "try_ros/src", and then catkin_make produced an error below. Not sure how to work around it.   Any idea?

```
-- ~~  traversing 4 packages in topological order:
...
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)
...
```

zhoulaifu, 2020-10-09:
In fact,  I could work around this problem with "catkin_make_isolated", but I am still interested in understanding why "catkin_make" does not work. Any clue?

dirk-thomas, 2020-10-09:
@zhoulaifu What branch / commit hash of this repository are you using?

zhoulaifu, 2020-10-16:
I got the following with "git log -1"

```
commit 91d2764ebd2344acf8e6754323f5b937b836d218 (HEAD -> noetic-devel, tag: 0.10.2, origin/noetic-devel, origin/HEAD)
Author: Dirk Thomas <dirk-thomas@users.noreply.github.com>
Date:   Fri Jul 31 15:21:32 2020 -0700

    0.10.2

```

dirk-thomas, 2020-10-16:
You error message

> `CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):`

 states that you are using ROS Melodic. Therefore you should use the `melodic-devel` branch of this repository (not `noetic-devel` as shown by your latest comment). That being said this isn't causing your problem and there must be something else in your workspace `try_ros/src`. Please share complete reproducible steps.

E.g. the following works just fine:

* `mkdir -p /tmp/ws/src`
* `git clone https://github.com/ros/ros_tutorials.git /tmp/ws/src/ros_tutorials`
* `cd /tmp/ws/`
* `. /opt/ros/noetic/setup.bash`
* `catkin_make`

Therefore I will go ahead and close this ticket since it is not a problem in this repository. Please feel free to continue commenting with the requested information. Maybe that allows to identify your problem.

closed 2020-10-16 by dirk-thomas

zhoulaifu, 2020-10-19:
Thanks for the detailed instructions!

I run these commands:

```
rm -rf /tmp/ws
mkdir -p /tmp/ws
git clone https://github.com/ros/ros_tutorials.git /tmp/ws/src/ros_tutorials 
cd /tmp/ws/src/ros_tutorials
git checkout --track origin/melodic-devel
cd /tmp/ws/
. /opt/ros/melodic/setup.bash
catkin_make

```
and got this error
```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/tmp/ws/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/ws/build/CMakeFiles/CMakeError.log".

```
Attached are the two log files referred to in the error message. 
[CMakeError.log](https://github.com/ros/ros_tutorials/files/5401403/CMakeError.log)
[CMakeOutput.log](https://github.com/ros/ros_tutorials/files/5401405/CMakeOutput.log)

dirk-thomas, 2020-10-19:
> I run these commands:

I just copied these commands into a Docker image (`docker run -i -t osrf/ros:melodic-desktop-full`) and it builds the four packages just fine.

> and got this error

The full output of the `catkin_make` invocation (before this error) should provide more details. Please copy-n-paste the full output without leaving out parts.

zhoulaifu, 2020-10-23:
Have not tried docker myself.  Here is what you asked -- the full outoput of the catkin_make

```
Base path: /tmp/ws
Source space: /tmp/ws/src
Build space: /tmp/ws/build
Devel space: /tmp/ws/devel
Install space: /tmp/ws/install
####
#### Running command: "cmake /tmp/ws/src -DCATKIN_DEVEL_PREFIX=/tmp/ws/devel -DCMAKE_INSTALL_PREFIX=/tmp/ws/install -G Unix Makefiles" in "/tmp/ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /tmp/ws/devel
-- Using CMAKE_PREFIX_PATH: /tmp/ws/devel_isolated/turtlesim;/tmp/ws/devel_isolated/rospy_tutorials;/tmp/ws/devel_isolated/roscpp_tutorials;/tmp/ws/devel_isolated/ros_tutorials;/opt/ros/melodic
-- This workspace overlays: /tmp/ws/devel_isolated/turtlesim;/tmp/ws/devel_isolated/rospy_tutorials;/tmp/ws/devel_isolated/roscpp_tutorials;/tmp/ws/devel_isolated/ros_tutorials;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /tmp/ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 4 packages in topological order:
-- ~~  - ros_tutorials (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'ros_tutorials' contains non standard content. Use the content of the following file instead: /tmp/ws/build/catkin_generated/metapackages/ros_tutorials/CMakeLists.txt
-- ~~  - roscpp_tutorials
-- ~~  - rospy_tutorials
-- ~~  - turtlesim
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/tmp/ws/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

zhoulaifu, 2020-10-23:
BTW, maybe I should not have asked the question -- is ROS1 still supported?

dirk-thomas, 2020-10-23:
> WARNING: The CMakeLists.txt of the metapackage 'ros_tutorials' contains non standard content. Use the content of the following file instead: /tmp/ws/build/catkin_generated/metapackages/ros_tutorials/CMakeLists.txt

Can you please share the content of these two files:
* `/tmp/ws/src/ros_tutorials/ros_tutorials/CMakeLists.txt`
* `/tmp/ws/build/catkin_generated/metapackages/ros_tutorials/CMakeLists.txt`

> is ROS1 still supported?

Yes, you can find the support dates for each ROS 1 distro in [REP 3](https://www.ros.org/reps/rep-0003.html). That being said they are no new ROS 1 distributions planned after Noetic.

zhoulaifu, 2020-11-10:
```
$ cat /tmp/ws/src/ros_tutorials/ros_tutorials/CMakeLists.txt
cmake_minimum_required(VERSION 3.0.2)
project(ros_tutorials)
find_package(catkin REQUIRED)
catkin_metapackage()
```

```
$ cat /tmp/ws/build/catkin_generated/metapackages/ros_tutorials/CMakeLists.txt
cmake_minimum_required(VERSION 2.8.3)
project(ros_tutorials)
find_package(catkin REQUIRED)
catkin_metapackage()
```

dirk-thomas, 2020-11-19:
@zhoulaifu The CMake error indicates that you are using an outdated version of `catkin_pkg`. As of ros-infrastructure/catkin_pkg#274 the above difference in the `VERSION` should be handled properly. Since you are using Melodic you need to make sure you have an up-to-date version of the Debian package `python-catkin-pkg-modules` (and no custom version of `catkin_pkg` installed via `pip`).

@sloretz @mabelzhang FYI