ros/ros_tutorials #110: turtlesim_node bug on slow computers

samlaf, 2020-10-21:
The turtlesim_node simulator updates the position of the turtle at a frequency of 62.5 hz (once every 16 ms).
I'm running it through docker on a macbook air 2015, which is so crappy the simulator falls behind, causing bugs.
For example, hitting the teleop once (forward arrow), causes a velocity of 2 unit/s to be sent to the simulator, and according to the semantics of the simulator (http://wiki.ros.org/turtlesim#line-324), this command is executed for 1 second (if no other command is sent during this time).
This means the turtle should end up 2units further after 1s. My turtle only moves by 0.8 units (varries, sometimes less sometimes a bit more, but never more than 1 unit).

sloretz, 2020-10-21:
What ROS distro are you using?

samlaf, 2020-10-22:
Originally kinetic, but I also tried on noetic and I get the same bug.
In turtle_frame.cpp (https://github.com/ros/ros_tutorials/blob/noetic-devel/turtlesim/src/turtle_frame.cpp)
I changed the line
`update_timer_->setInterval(16);`
to
`update_timer_->setInterval(100);`

This lowers the update of the simulator to 10 hz.
The rotations are a bit choppier but at least this way forward velocity commands move the turtle the entire 2units.

sloretz, 2020-10-23:
I'm surprised it's that slow on a 2015 macbook air. The `turtle_frame.cpp` code hasn't changed that much since the ROS Indigo release in 2014.

https://github.com/ros/ros_tutorials/compare/indigo-devel..noetic-devel#diff-ac97e1355dfb24394aba2ddae77bf89b072cdc21bb55adbea6ed939fc553b025

Any idea what part of the update is the most expensive on that machine?

samlaf, 2020-10-27:
I'm honestly not sure what's causing the problem anymore... I just checked `docker stats` and its saying I'm barely using 3% CPU...
[image]

Could it be a bug of docker on mac?

sloretz, 2020-10-27:
> Could it be a bug of docker on mac?

Maybe, or a GPU issue? Maybe software is unable to use the GPU from inside the docker container? Can it run any other GUI programs from inside the container - like RViz?

samlaf, 2020-10-30:
Does the simulator use a GPU??
I can run rqt fine, but RViz gives me this error:

> [ INFO] [1604034370.878219200]: rviz version 1.12.17
> [ INFO] [1604034370.879726800]: compiled against Qt version 5.5.1
> [ INFO] [1604034370.880033800]: compiled against OGRE version 1.9.0 (Ghadamon)
> process 101: D-Bus library appears to be incorrectly set up; failed to read machine uuid: UUID file '/etc/machine-id' should contain a hex string of length 32, not length 0, with no other text
> See the manual page for dbus-uuidgen to correct this issue.
> libGL error: No matching fbConfigs or visuals found
> libGL error: failed to load driver: swrast
> libGL error: No matching fbConfigs or visuals found
> libGL error: failed to load driver: swrast
> libGL error: No matching fbConfigs or visuals found
> libGL error: failed to load driver: swrast
> rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/OgreGLSupport.cpp:56: virtual void Ogre::GLSupport::initialiseExtensions(): Assertion `pcVer && "Problems getting GL version string using glGetString"' failed.
> Aborted

sloretz, 2020-11-02:
> Does the simulator use a GPU??

Hmm, I'm not sure, but maybe. It might be that QT is trying to render using the CPU because it can't access the GPU, and that's really slow. If I understand correctly, the RViz error message means OGRE is trying and failing to use software rendering instead of the GPU. Does `glxgears` work inside the container?

It doesn't seem like there's something that can be fixed in this repo, so I'll close this issue for now. I would look at trying to create a docker container that can use the GPU on a mac. 

https://stackoverflow.com/questions/46367517/can-i-use-my-gpu-from-a-docker-container-on-a-macbook-pro-amd-radeon-gpu
https://askubuntu.com/questions/834254/steam-libgl-error-no-matching-fbconfigs-or-visuals-found-libgl-error-failed-t

closed 2020-11-02 by sloretz