ros/ros_tutorials #115: Incorrect shebang for Python script with Noetic

BenjaminNavarro, 2021-03-12:
With Noetic, ROS has switched to Python 3 but the scripts in the tutorials still refer to `python`in their shebang:
```
#!/usr/bin/env python
```
but `python` doesn't exist on Ubuntu 20.04 so the script will just fail to run with:
```
/usr/bin/env: ‘python’: No such file or directory
```
Starting from Noetic the proper shebang is:
```
#!/usr/bin/env python3
```
This should be fixed in this repo and in the wiki

mabelzhang, 2021-03-12:
This is actually by design. We document the python3 shebang specific to ROS here http://wiki.ros.org/UsingPython3/SourceCodeChanges#Changing_shebangs

`python` is the correct shebang. `catkin_install_python`, which is called from `CMakeLists.txt`, will automatically append the `3` as needed.

mabelzhang, 2021-03-16:
Closing as I think it has been answered. Feel free to reopen.

closed 2021-03-16 by mabelzhang

BenjaminNavarro, 2021-03-19:
Ok, I didn't know, probably because I mostly use C++ and not Python