ros/ros_tutorials #120: Missing rolling-devel branch

hidmic, 2021-04-23:
There's a tutorial in Rolling that expects the `rolling-devel` branch to exist here, but there's none. We could drop that reference, but should there be a `rolling-devel` branch? I see no `master` nor `ros2` branch either. It appears we've been creating distro specific branches as we go here (e.g. see #100).

hidmic, 2021-04-23:
FYI @sloretz.

sloretz, 2021-04-23:
I'd be fine with any of `rolling-devel`, `master` or `main`. @mabelzhang Any preferences on branch name?

mabelzhang, 2021-04-26:
`rolling-devel` or `main` or `ros2` is fine. For an external contributor, maybe `ros2` is the most intuitive to create PRs? `main` could be ambiguous whether it's ROS 1 or 2.

hidmic, 2021-06-07:
> For an external contributor, maybe `ros2` is the most intuitive to create PRs

Don't have a strong opinion. But I do think we need to branch and get that tutorial fixed.

mabelzhang, 2021-06-07:
Looks like the one in common above is `rolling-devel`, which is also the expected branch. Feel free to create it -- or I can do that if you want. I guess it'd branch off of `galactic-devel`?

hidmic, 2021-06-07:
Just added `rolling-devel`, branching off of `galactic-devel`.

closed 2021-06-07 by hidmic