ros/ros_tutorials #126: [ROS2 galactic windows] cannot start turtlesim - no Qt platform plugin could be initialized

flynneva, 2021-06-18:
ive followed the ros2 galactic installation instructions (installed 6.2.0 instead of 5.15.0) and set the proper env vars as stated in the instructions for windows.

I've confirmed that the `qwindows.dll` exists in the path below when I copy & paste it into a finder window....just it seems I am missing something else.

Any advice would be extremely helpful!

```
 ros2 run turtlesim turtlesim_node
qt.qpa.plugin: Could not find the Qt platform plugin "windows" in "C:\Qt\6.2.0\msvc2019_64\plugins\platforms"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

```

flynneva, 2021-06-18:
found this on stack overflow that sounds very similar to my problem: https://stackoverflow.com/questions/20495620/qt-5-1-1-application-failed-to-start-because-platform-plugin-windows-is-missi

flynneva, 2021-06-18:
going to install 5.15.0 and see if that fixes the problem

flynneva, 2021-06-18:
same problem with Qt 5.15.0 on windows.

```
 ros2 run turtlesim turtlesim_node
qt.qpa.plugin: Could not find the Qt platform plugin "windows" in "C:\Qt\5.15.0\msvc2019_64\plugins\platforms"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

```

Ehsan8994, 2021-08-13:
I have same problem, this work for me maybe solve your problem. copy "qwindows.dll" from C:\dev\ros2_galactic\bin\platforms and replace it with the same in "C:\Qt\5.15.0\msvc2019_64\plugins\platforms".

icsl-Jeon, 2021-08-16:
> I have same problem, this work for me maybe solve your problem. copy "qwindows.dll" from C:\dev\ros2_galactic\bin\platforms and replace it with the same in "C:\Qt\5.15.0\msvc2019_64\plugins\platforms".

It works! thank you

jonathan-sung, 2021-12-24:
> I have same problem, this work for me maybe solve your problem. copy "qwindows.dll" from C:\dev\ros2_galactic\bin\platforms and replace it with the same in "C:\Qt\5.15.0\msvc2019_64\plugins\platforms".

It worked! Thank you so much!

sloretz, 2022-05-11:
Installing Qt 5.12.12 resolves the issue for me. I opened a PR to update the ROS 2 install instructions: https://github.com/ros2/ros2_documentation/pull/2499

closed 2022-05-13 by sloretz