ros/ros_tutorials #127: Adding holonomic motion disable (differential drive motion only) option to turtlesim

Tiryoh, 2021-06-29:
> I just find out that the turtle is holonomic now.  Because I am teaching ROS these years so want to share some thoughts here though this issue is closed.
> 
> Introducing the holonomic movement into Turtlesim is great, it does not change the learning experience. And I also want this feature to test some easy multi-robot formation or cooperation stuff. **However, it brings in some confusion during learning TF using turtle_tf.** My students now solve the problem mainly based on the global info, the relation between two turtles can be ignored and you can command the follower based on v.x and v.y with a zero-valued a.z. A differential model may indicate that the follower must perceive the target turtle.
> 
> Moreover, the kinematic model of the turtle is no longer the same as a Turtlebot. 
> 
> It won't be a problem during our class. I don't know how this will affect the self-learner. I think your intuition is considerable @dirk-thomas .  I prefer a default differential turtle and can be switched to omni-directional. 
> 
> best regards,

_Originally posted by @tianb03 in https://github.com/ros/ros_tutorials/issues/93#issuecomment-869282435_

I created the issue #93 and the related PR because I believe it will be useful in introducing beginners that it is possible to work with holonomic motion.
However, it seems that if this option is always enabled, it may sometimes disturb the learning process for beginning students.
This is not what I want to do.
It would be nice to have an option to disable holonomic motion as @jacobperron mentions.
I'm going to submit a pull request for the option to disable it.
How do you think the options should be able to be specified, @tianb03, @jacobperron?
I believe that rosparam is easier to specifying both from the command line and the roslaunch like the [teleop_twist_keyboard](https://github.com/ros-teleop/teleop_twist_keyboard).

e.g.)

```sh
rosrun turtlesim turtlesim_node _holonomic:=false
```

```xml
<arg name="holonomic" default="false" />
<node pkg="turtlesim" name="sim" type="turtlesim_node">
  <param name="holonomic" value="$(arg false)" />
</node>
```

tianb03, 2021-07-05:
Thanks for your efforts @Tiryoh , using rosparam is great!

jacobperron, 2021-07-07:
Using a ROS parameter to disable holonomic motions sounds okay to me.

Tiryoh, 2022-06-25:
I'm closing this issue because the PR has been merged to the rolling branch.

https://github.com/ros/ros_tutorials/pull/131

Thanks @tianb03, @jacobperron, @audrow!

closed 2022-06-25 by Tiryoh