ros/ros_tutorials #135: Error in teleop_turtle_key.cpp "E0349 no operator "=" matches these operands"

bobleeswaggs, 2021-10-13:
Was having failures with `colcon build --merge-install` for the tutorial workspace build. After finding the source of the first error that shows up under inspection, the error appears in the file of the title, on line 318. I pulled this from the repo, is there any fix to this? This error is causing the colcon build to fail.

tlalexander, 2022-01-25:
I get this error too. It looks like the error has moved to line 323:

```
/home/taylor/Software/dev_ws/src/ros_tutorials/turtlesim/tutorials/teleop_turtle_key.cpp:323:7: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<turtlesim::action::RotateAbsolute>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<turtlesim::action::RotateAbsolute> > >)>’} and ‘TeleopTurtle::sendGoal(float)::<lambda(rclcpp_action::ClientGoalHandle<turtlesim::action::RotateAbsolute>::SharedPtr)>’)
  323 |       };
```

audrow, 2022-01-26:
Hi @bobleeswaggs and @tlalexander, a few questions to help us find the bug:

- What ROS distro are you using?
- What operating system are you using?
- Are you building from source or using binaries?

tlalexander, 2022-01-27:
Hi @audrow thanks for the reply.
It turns out I had been using foxy but was trying to build the code for galactic. I've installed galactic and now the code builds. So my mistake!
Since the code builds, perhaps this issue could be closed.
Thanks!

audrow, 2022-01-27:
@tlalexander, okay thanks for the update.

closed 2025-10-17 by ahcorde