ros/ros_tutorials #138: turtlesim without graphical frontend

lepalom, 2022-03-23:
I have not found it, but I would like to ask if it would be possible to execute turtlesim_node without graphical frontend. 

It sounds strange and let me explain a bit more.  I'm giving a ROS course and I would like to review the exercises of my students in a CI script, and AFAIK I cannot execute (or have sense) turtlesim_node without graphical (or can I?). So, just to test if a service or topic is created in a turtlesim client I need to execute it. If I could run turtlesim_node in some text mode, or just some kind of blind mode, I could then run some test to check it.

Is it possible?

audrow, 2022-03-23:
I'm not sure. It seems like the turtlesim visualization is pretty tightly coupled with the program logic. I would bet that you can comment out the visualization without too much effort.

It might be nice to add a parameter to turtlesim that can disable visualization. I'm going to mark this as help wanted. @lepalom, if you'd like to make a PR for this, feel free to @ me for a review.

lepalom, 2022-03-24:
@audrow I have checked the sources and I guess that it can be possible. I cannot do it now, but maybe in summer I can .

vovo-4K, 2025-02-13:
@lepalom The turtlesim_node can be launched in headless mode using the platform offscreen option:
```
ros2 run turtlesim turtlesim_node -platform offscreen
```

sloretz, 2025-02-13:
Thanks @vovo-4K ! Closing as the question is answered

closed 2025-02-13 by sloretz