ros/ros_tutorials #144: README file?

ppbrown, 2022-09-19:
could we please have a top level README file for how to build these things?
I'm completely new to ros and catkin. I learn things best by seeing them work.
But I dont know how to get the build process to work here :(

audrow, 2022-09-19:
We have the documentation elsewhere.

Here's the documentation for building on ROS 1 (you can also find other tutorials [here](http://wiki.ros.org/ROS/Tutorials)):
http://wiki.ros.org/ROS/Tutorials/BuildingPackages

Here's documentation for ROS 2:
https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html

I'm going to mark this as help-wanted since it would be nice to have this, even though this info is covered in tutorials.

ppbrown, 2022-09-19:
thats fair... but how about just putting the info you put right here, into a top level readme file?

ppbrown, 2022-09-19:
although.. are both those links relevant? is the tutorial actually equally valid for both ROS1 and ROS2?

audrow, 2022-09-30:
Somewhat confusingly, this repo has both ROS 1 and ROS 2 branches, so both would apply to their corresponding branches.