ros/ros_tutorials #147: colcon build error

XZ-Niupitang, 2023-03-03:
/ROS/src/ros_tutorials/turtlesim/tutorials/teleop_turtle_key.cpp:200:5: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<turtlesim::action::RotateAbsolute>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<turtlesim::action::RotateAbsolute> >)>’} and ‘TeleopTurtle::sendGoal(float)::<lambda(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<turtlesim::action::RotateAbsolute> > >)>’)
  200 |     };
      |     ^
After the error is reported here, teleop_turtle_key.cpp is canceled the sendGoal () function in the compilation is normal, what is the problem?

```
void TeleopTurtle::sendGoal(float theta)
{
  auto goal = turtlesim::action::RotateAbsolute::Goal();
  goal.theta = theta;
  auto send_goal_options = rclcpp_action::Client<turtlesim::action::RotateAbsolute>::SendGoalOptions();
  // send_goal_options.goal_response_callback =
  //   [this](std::shared_future<rclcpp_action::ClientGoalHandle<turtlesim::action::RotateAbsolute>::SharedPtr> future)
  //   {
  //     RCLCPP_DEBUG(nh_->get_logger(), "Goal response received");
  //     this->goal_handle_ = future.get();
  //   };
  rotate_absolute_client_->async_send_goal(goal, send_goal_options);
}
```

mjcarroll, 2023-03-03:
We are going to need more information about which platform and versions you are using.

XZ-Niupitang, 2023-03-04:
> 我们将需要有关您正在使用的平台和版本的更多信息。

The system version is Ubuntu 22.04LTS，The ROS2 environment is humble，After logging out, it is running normally. Creating a package or node under the workspace can compile and work normally. I just want to know whether the code I logged out will cause ROS2 to run incorrectly.

closed 2025-10-17 by ahcorde